hartkopp/can-isotp #2: isotp.c: can_rx_register() kernel version guard is incorrect

barsnick, 2017-08-01:
With Linux 4.4.67, I get this error:

```
net/can/isotp.c: In function ‘isotp_bind’:
net/can/isotp.c:1120:2: error: too few arguments to function ‘can_rx_register’
  can_rx_register(dev, addr->can_addr.tp.rx_id,
  ^
In file included from net/can/isotp.c:67:0:
include/linux/can/core.h:48:5: note: declared here
 int can_rx_register(struct net_device *dev, canid_t can_id, canid_t mask,
     ^
scripts/Makefile.build:258: recipe for target 'net/can/isotp.o' failed
make[2]: *** [net/can/isotp.o] Error 1
```

Apparently, the kernel API of can_rx_register() was changed in 4.4.50, not 4.9.11. I commented in downwith's repo here:

https://github.com/downwith/linux-isotp/commit/d3bb87f6cda11d6cad3be80752f402190845c44a#commitcomment-23411680

The upstream commit which changed it was this:
https://github.com/torvalds/linux/commit/e6c654c9c09258dc0f82d1baa9ce69aa68bc735e

Too bad both you and downwith chose to copy over the code (that's so 20th century ;-)), instead of doing github-internal forks. With the latter solution, we could have tracked history through all of the development.

downwith, 2017-08-01:
Also, Github is so 2010 😎

barsnick, 2017-08-01:
My assessment was incorrect: The change was in 4.10(-rc7), apparently backported to 4.9 (LTS) in 4.9.11, and also backported to 4.4 (LTS) in 4.4.50.

I'll try to come up with a proper version check, unless you beat me to it.

hartkopp, 2017-08-02:
Fixed :-)

Tnx Moritz!

See: https://github.com/barsnick/linux/commit/0fc57db43ba0daead71c8b534ba81f5d17c2ee1a

closed 2017-08-02 by hartkopp