hartkopp/can-isotp #7: sign-file: certs/signing_key.pem: No such file or directory

akast, 2017-10-22:
akast@ubuntu:~/can-isotp$ sudo make modules_install
make -C /lib/modules/4.10.0-19-generic/build M=/home/akast/can-isotp modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-19-generic'
  INSTALL /home/akast/can-isotp/net/can/can-isotp.ko
At main.c:158:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.10.0-19-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-19-generic'

hartkopp, 2017-10-22:
This is pretty much like this;
https://github.com/patjak/bcwc_pcie/issues/70#issuecomment-214289165

Can you still load and run the module?

Don't know about module signing and its effects to out-of-tree modules.

On my machine loading can-isotp only results in:
can_isotp: loading out-of-tree module taints kernel.
can: isotp protocol (rev 20170725 alpha)

hartkopp, 2017-11-06:
Closing due to missing feedback.

closed 2017-11-06 by hartkopp

AndySchroder, 2020-02-15:
I am running ubuntu 18.04 and had the same problem, except the SSL error lines were slightly different. Generating a signing key and putting it in the appropriate certs folder then allowed the module to be signed. If you don't create the signing key, the module can be run from the compilation folder, but I don't think installed.