hartkopp/can-isotp #9: Not able to compile on raspberry pi jessie

swanandb2, 2018-01-27:
I did tried to compile on raspberry pi jessie and observed following error:

make -C **/lib/modules/4.9.78-v7+/build** M=/home/pi/socCan/can-isotp PROJECT_DIR=/home/pi/socCan/can-isotp modules
make[1]: *** /lib/modules/4.9.78-v7+/build: No such file or directory.  Stop.
Makefile:21: recipe for target 'modules' failed
make: *** [modules] Error 2

I did tried to fix this using apt update and upgrade but issue is not getting fixed.

Following is my system configuration
Linux pilinux 4.9.78-v7+ #1084 SMP Thu Jan 25 18:05:49 GMT 2018 armv7l GNU/Linux

Can someone please help me to resolve this issue?

hartkopp, 2018-01-27:
Please follow https://github.com/hartkopp/can-isotp/blob/master/README.isotp !
At least you missed to install the correct linux-headers package.
Before installing the kernel version specific Linux headers I had the same error message too.

swanandb2, 2018-01-29:
[issue.txt](https://github.com/hartkopp/can-isotp/files/1674113/issue.txt)
I did tried to follow steps instruction looks like headers 4.9.78V are not present on raspbian cache.
I feel only way is to re-install raspbian OS to 4.9.68V as I did not find any headers for 4.9.78. Please refer attach file for more details.

hartkopp, 2018-01-29:
E: Unable to locate package linux-headers-4.9.78-v7

Is there a '+' missing?

Did you try 
apt-get install linux-headers-4.9.78-v7+
??

Can you send the error message when trying to load the can-isotp.ko module?
Did you load the CAN framework module, e.g. with 'modprobe can' before?

FastDeath, 2018-02-22:
On a raspberrypi you should use ``rpi-source`` (refer to https://github.com/notro/rpi-source/wiki) to download the linux source code to something like ``./linux-<some hash>``. Usually though, the source is located under ``/lib/modules/<kernel version>/build`` so you need to copy/move or symlink to the just downloaded folder this ``build`` directory (or to ``/usr/src/..`` and symlink to there).
I used the following command line to do this:
```
ln -s /root/linux-90ac037dbeecbb514b677e43b53bc5b1a452ea22/ /usr/lib/modules/`uname -r`/build
```
Now you should be able to ``make`` the module, which you can then copy to the appropriate folder to load it using modprobe or automatically (``/lib/modules/`uname -r`/extra/net/can``). After copying run ``depmod`` to rebuild the kernel module maps.

Hope this helps! :)

(Tested on arch linux / kernel version 4.14.18-v7+)

swanandb2, 2018-02-23:
Thanks for your suggestion. I was able to build source code now

closed 2018-02-23 by swanandb2