hartkopp/can-isotp #11: Requires a way to reset socket state

pylessard, 2018-12-14:
This issue is a mix of requirements and questions that arise maybe from my limited knowledge of the Linux kernel. I have been facing many times in the past a situation where I open many IsoTP socket and don't close them properly and end up not being able to use a specific socket address. 

Basically, I try to create a new socket using the same txid and rxid of the previous unclosed socket and the result is that I cannot changes its parameters (stmin, bs, wftmax) and reception is working interrmitently, (just like if 2 sockets were emptying the same buffer).

I know I created that problem by mishandling the sockets, but when that happens, I have found no other solution than rebooting my mahcine to clean that up. I've tried resetting the networking services, putting down/up the vcan interface, unloading the kernel module. None works.

Is there a way to reset the state of the kernel module?

pylessard, 2018-12-15:
I understand that this can't work. The goal here is to delete the previous socket before opening the new one. Seems that in some case, I can't do that. But as I said, I can't tell exactly how I get in this situation, so I believe this should be closed.

rmmod return "module is busy", whuch let me believe something is hanging in a undefined state.

Will provide details once I do have them

hartkopp, 2018-12-16:
"module is busy" would only occur when there's still an open socket handled by that module.
Closing this issue for now. Tnx!

closed 2018-12-16 by hartkopp