hartkopp/can-isotp #13: Incorrect include path during kernel build?

awilke, 2019-02-08:
I am trying to create a `deb` package for this module. I can successfully create the package, but I get this compilation error when trying to install it:
```
gcc -Wp,-MD,/var/lib/dkms/can-isotp/0.1.0/build/net/can/.isotp.o.d -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/4.8/include -I./arch/x86/include -Iarch/x86/include/generated/uapi -Iarch/x86/include/generated -Iinclude -I./arch/x86/include/uapi -Iarch/x86/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -Iubuntu/include -D__KERNEL__ -fno-pie -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -fno-pie -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mtune=generic -mno-red-zone -mcmodel=kernel -funit-at-a-time -maccumulate-outgoing-args -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_FXSAVEQ=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_CRC32=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -DRETPOLINE -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -O2 --param=allow-store-data-races=0 -DCC_HAVE_ASM_GOTO -Wframe-larger-than=1024 -fstack-protector -Wno-unused-but-set-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -pg -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -I/include/uapi -DMODULE '-DKBUILD_STR(s)=#s' '-DKBUILD_BASENAME=KBUILD_STR(isotp)' '-DKBUILD_MODNAME=KBUILD_STR(can_isotp)' -c -o /var/lib/dkms/can-isotp/0.1.0/build/net/can/.tmp_isotp.o /var/lib/dkms/can-isotp/0.1.0/build/net/can/isotp.c
/var/lib/dkms/can-isotp/0.1.0/build/net/can/isotp.c:74:29: fatal error: linux/can/isotp.h: No such file or directory
 #include <linux/can/isotp.h>
                             ^
compilation terminated.
make[2]: *** [/var/lib/dkms/can-isotp/0.1.0/build/net/can/isotp.o] Error 1
make[1]: *** [/var/lib/dkms/can-isotp/0.1.0/build/net/can] Error 2
make: *** [_module_/var/lib/dkms/can-isotp/0.1.0/build] Error 2
make: Leaving directory `/usr/src/linux-headers-4.4.0-116-generic'
```
Based on `-I/include/uapi` in the `gcc` command, I believe that the failure is because `PROJECT_DIR` is unset in `net/can/Makefile`. Presumably this is because the kernel's `Makefile` does not propagate `PROJECT_DIR` during the second-pass invocation of `can-isotp`'s top-level `Makefile`...?

The package installs correctly if I change `net/can/Makefile` to use this instead:
```
ccflags-y += -I$(M)/include/uapi
```
However, I am unfamiliar with best practices so I'm not sure whether this is the best fix. Suggestions?

hartkopp, 2019-02-08:
Can you check if the patch above fixes your problem?

Thanks

awilke, 2019-02-08:
Yup, that patch works great!

hartkopp, 2019-02-09:
Thanks for testing!

closed 2019-02-09 by hartkopp