hartkopp/can-isotp #14: Error while building

MaterFertex, 2019-05-24:
I run the make command and this is the answer that I receive is:

make -C /lib/modules/3.16.0-8-amd64/build M=/home/rubio/can-isotp modules
make[1]: Entering directory '/usr/src/linux-headers-3.16.0-8-amd64'
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-8-amd64'
  CC [M]  /home/rubio/can-isotp/net/can/isotp.o
/home/rubio/can-isotp/net/can/isotp.c: In function ‘isotp_bind’:
/home/rubio/can-isotp/net/can/isotp.c:1141:2: error: too few arguments to function ‘can_rx_register’
  can_rx_register(dev, addr->can_addr.tp.rx_id,
  ^
In file included from /home/rubio/can-isotp/net/can/isotp.c:67:0:
/usr/src/linux-headers-3.16.0-8-common/include/linux/can/core.h:48:5: note: declared here
 int can_rx_register(struct net_device *dev, canid_t can_id, canid_t mask,
     ^
/usr/src/linux-headers-3.16.0-8-common/scripts/Makefile.build:262: recipe for target '/home/rubio/can-isotp/net/can/isotp.o' failed
make[5]: *** [/home/rubio/can-isotp/net/can/isotp.o] Error 1
/usr/src/linux-headers-3.16.0-8-common/scripts/Makefile.build:409: recipe for target '/home/rubio/can-isotp/net/can' failed
make[4]: *** [/home/rubio/can-isotp/net/can] Error 2
/usr/src/linux-headers-3.16.0-8-common/Makefile:1362: recipe for target '_module_/home/rubio/can-isotp' failed
make[3]: *** [_module_/home/rubio/can-isotp] Error 2
Makefile:181: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-8-amd64'
Makefile:21: recipe for target 'modules' failed
make: *** [modules] Error 2

I have tried to upgrade my kernel and also y update can-utils but I still receiving the same error.

hartkopp, 2019-05-24:
There was a fix backported to Linux 3.18 LTS.
Updated the kernel version guard to catch the 3.18 LTS kernel.
Please pull and try again!

Tnx!

hartkopp, 2019-05-24:
Arg! You had a 3.16 - and not a 3.18 ... will apply a patch for that in a minute.
Stay tuned.

hartkopp, 2019-05-24:
Done

MaterFertex, 2019-05-24:
Thanks! Now is working!

closed 2019-05-24 by MaterFertex

swanandb2, 2019-08-13:
I did tested with sudo apt-get install build-essential linux-headers command and it did work but now
for make command it throws following error ( build directory is missing)
pi@raspberrypi:~/uds/can-isotp $ sudo make
make -C /lib/modules/4.19.57-v7+/build M=/home/pi/uds/can-isotp modules
make[1]: Entering directory '/lib/modules/4.19.57-v7+/build'
make[1]: *** No rule to make target 'modules'. Stop.
make[1]: Leaving directory '/lib/modules/4.19.57-v7+/build'
make: *** [Makefile:21: modules] Error 2

Following are content of /lib/modules/4.19.57-v7+/ directory

pi@raspberrypi:/lib/modules/4.19.57-v7+ $ ls -l
total 2092
drwxr-xr-x 11 root root 4096 Jul 10 05:37 kernel
-rw-r--r-- 1 root root 525300 Jul 8 18:32 modules.alias
-rw-r--r-- 1 root root 546338 Jul 8 18:32 modules.alias.bin
-rw-r--r-- 1 root root 11287 Jul 8 18:33 modules.builtin
-rw-r--r-- 1 root root 12386 Jul 8 18:33 modules.builtin.bin
-rw-r--r-- 1 root root 180509 Jul 8 18:32 modules.dep
-rw-r--r-- 1 root root 250914 Jul 8 18:32 modules.dep.bin
-rw-r--r-- 1 root root 302 Jul 8 18:33 modules.devname
-rw-r--r-- 1 root root 61698 Jul 8 18:32 modules.order
-rw-r--r-- 1 root root 352 Jul 8 18:33 modules.softdep
-rw-r--r-- 1 root root 231312 Jul 8 18:32 modules.symbols
-rw-r--r-- 1 root root 284469 Jul 8 18:32 modules.symbols.bin

Can you please help me for direction? Let me know if you need any other details.
pi@raspberrypi:/lib/modules/4.19.57-v7+ $ uname -a
Linux raspberrypi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

hartkopp, 2019-08-13:
I wonder what the directory '/lib/modules/4.19.57-v7+/build' really contains on your system. Usually this symlink points to the complete linux kernel source that was used to build the modules - and containing all the include files / configs and arch specific configurations.

Please check the Raspi community for compiling out-of-tree modules.