hartkopp/can-isotp #15: warning when make modules_install

hsilva7, 2019-06-20:
I followed the instructions to install the module, in which I get a warning and the module isn't installed.

I am on a beagle bone black with kernel 4.19.24-ti-r10

this is what I get after running `sudo make modules install`

```

make -C /lib/modules/4.19.24-ti-r10/build M=/home/debian/can-isotp modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.19.24-ti-r10'
  INSTALL /home/debian/can-isotp/net/can/can-isotp.ko
  DEPMOD  4.19.24-ti-r10
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.24-ti-r10'
```

hartkopp, 2019-06-20:
Hm - these are standard tools that usually find a System.map in the /boot directory.

Don't know where the BBB hosts its System.map

Please check where to find that file and refer it when invoking "depmod" by hand. E.g. "depmod -a -F /boot/System.map-4.19.24-ti-r10"

hsilva7, 2019-06-25:
Not sure what you mean by running by hand...
I have run `depmod -a -F /boot/System.map-4.19.24-ti-r10` and the file does exist. The command just executes and exit without any result.

hartkopp, 2019-06-26:
Ok, that should have worked then - and `modprobe can-isotp` should run without problems. 

Usually the sytem map is located in /lib/modules/<kernelversion>/ therefore the default process obviously failed probably as you just worked on base of the linux-headers.

hsilva7, 2019-06-26:
Ok, thank you.

Assuming that everything is right, I am trying to run 
` echo "45 69 6e 65 20 73 65 68 72 20 6c 61 6e 67 65 20 4e 61 63 68 72 69 63 68 74" | isotpsend -s 321 -d 123 vcan0`
and 
` isotpdump -c -a -s 321 -d 123 vcan0`
I expected to get 
`vcan0  321  [8]  [FF] ln: 25   data: 45 69 6E 65 20 73     -  'Eine s'
vcan0  123  [3]  [FC] FC: 0 = CTS # BS: 0 = off # STmin: 0x00 = 0 ms
vcan0  321  [8]  [CF] sn: 1    data: 65 68 72 20 6C 61 6E  -  'ehr lan'
vcan0  321  [8]  [CF] sn: 2    data: 67 65 20 4E 61 63 68  -  'ge Nach'
vcan0  321  [6]  [CF] sn: 3    data: 72 69 63 68 74        -  'richt'`

but I only get
`can0  321  [8]  [FF] ln: 25   data: 45 69 6E 65 20 73     -  'Eine s'`

hartkopp, 2019-06-26:
Yes. You need a isotprecv to receive the data stream too.

hartkopp, 2019-08-09:
Seems to be fixed ...

litaoxyz, 2020-02-16:
> Ok, thank you.
>
> Assuming that everything is right, I am trying to run
> ` echo "45 69 6e 65 20 73 65 68 72 20 6c 61 6e 67 65 20 4e 61 63 68 72 69 63 68 74" | isotpsend -s 321 -d 123 vcan0`
> and
> ` [...]

I have the same problem with you. how did you solved?

hartkopp, 2020-02-16:
isotpsend and isotpdump are not sufficient to perform a proper isotp data transfer.
Your need isotpsend AND isotprecv  as already stated here https://github.com/hartkopp/can-isotp/issues/15#issuecomment-505869073 .
isotpdump ist just to dump contents (read-only) - so there's no communication counterpart, like isotprecv.