hartkopp/can-isotp #16: SIOCGSTAMP fails with Linux Kernel 5.2.1

akorb, 2019-07-23:
Recently, I updated to the kernel `5.2.1-1-MANJARO`.

Since then ISOTP socket throw an error. With `5.1.18-1-MANJARO` the error does not appear.

Program to reproduce:
```c
#include <sys/types.h>
#include <sys/socket.h>
#include <sys/ioctl.h>
#include <errno.h>

#include <sys/time.h>
#include <stdio.h>
#include <unistd.h>

#define CAN_ISOTP 6

int main() {
    int fd = socket(AF_CAN, SOCK_DGRAM, CAN_ISOTP);

    struct timeval tv;
    int res = ioctl(fd, SIOCGSTAMP, &tv);

    printf("error code: %d\nerrorno: %d\n", res, errno);
    perror("ioctl() failed");

    close(fd);
    return 0;
}
```

**Output:**
```
~ $ ./test
error code: -1
errorno: 25
ioctl() failed: Inappropriate ioctl for device
```

For `5.1.18-1-MANJARO` the output is:

Which is fine because no packet has been received.

This commit might be related to this problem:
https://github.com/torvalds/linux/commit/c7cbdbf29f488a19982cd9f4a109887f18028bbb#diff-35a396ec1d0c44b5f7a45af70ec25268

hartkopp, 2019-07-23:
Hi Andreas! Thanks for the report.
Can you please test the latest version?
Thanks,
Oliver

akorb, 2019-07-23:
Thanks for the fast reply and fix.
It works again.

closed 2019-07-23 by akorb