hartkopp/can-isotp #17: Would you consider getting this in Debian repositories?

brandonros, 2019-11-07:


hartkopp, 2019-11-07:
Hm - no.
I consider to bring this upstream. After the presentation of CAN_J1939 at Embedded Linux Conference Europe last month, I feel confident to send this on the mailing list too :-)

brandonros, 2019-11-07:
ah yes, that is the better option.

I think it will land nicely, it is very good work. Thank you for all of your efforts!!!

closed 2019-11-07 by brandonros

hartkopp, 2019-11-07:
Are you subscribed to the linux-can mailing list?
If you are interested in testing the upstream version it would be coll to stay in touch there.
Are you using the master branch of can-isotp or e.g. the softhrt branch (https://github.com/hartkopp/can-isotp/commits/softhrt) I tend to post for mainline inclusion.