hartkopp/can-isotp #18: Doubt: Message Recovery

alfonsobrandao, 2019-12-18:
Hey everyone!

I have a simple can client server, that uses can-isotp to send data, and I want to know if there's any way of finding if the message was fully sent, if there's any packet loss, or any sort of callback that helps me lead with message recovery.

Thanks is advance!
Alfonso

hartkopp, 2019-12-18:
Hello Alfonso,
by default ISO-TP is an unreliable protocol like UDP.
When the iso-tp implementation sends CAN frames to the CAN driver layer - which can potential fail due to a tx queue len overflow. Therefore you should increase the tx-queue-len as decribed in https://github.com/hartkopp/can-isotp/blob/master/README.isotp

The J1939 implementation checks whether it was able to pass the CAN frame to the network driver layer - and does a retry if it fails. This should be added to iso-tp too.

Successfully sent CAN frames are echo'ed back into the system. You can see them e.g. with a CAN_RAW socket. But iso-tp and J1939 do not check the successful sending of each frame on the bus today.

alfonsobrandao, 2019-12-18:
Hi Hartkopp!

Thank you very much for your fast and detailed answer :) 

Best regards,
Alfonso

closed 2019-12-18 by alfonsobrandao