hartkopp/can-isotp #20: isotpsniffer "socket: Protocol not supported"

granmadave, 2020-01-05:
I am convinced that I installed and made can-isotp properly, but for some reason, I'm not able to rock with isotpsniffer.  Please advise.

hartkopp, 2020-01-06:
Did you try `insmod can-iso.tp` by hand after building the source code ?

granmadave, 2020-01-06:
Hmm.  no.  I hadn't seen that step, I don't think.  I'll give it a try!!

Thanks!

granmadave, 2020-01-07:
I cannot find that file; I receive 
insmod: ERROR: could not load module can-iso.tp: No such file or directory

hartkopp, 2020-01-07:
Oops! My fault. It had to be `insmod can-isotp.ko`

Btw. when you followed the receipt at
https://github.com/hartkopp/can-isotp/blob/master/README.isotp
everything should be fine at first sight. Or did you get any errors when building the kernel module following the README?

granmadave, 2020-01-07:
I thought everything was hip, including doing the insmod of can-isotp.ko,
the whole README... everything.

I'll go home and verify.

granmadave, 2020-01-07:
I was so sure that I had already done this, but I went ahead and did it again: 

sudo insmod ./net/can/can-isotp.ko

but I receive a "insmod: ERROR: could not insert module ./net/can/can-isotpko: Invalid module format"

So, I went through and retraced my steps, thinking I must have done something incorrectly.  When I went for the "make", isotp.c gives me a fatal error: linux/can/isotp.h: No such file or directory
#include <linux/can/isotp.h> (with pink!)
compilation terminated.

followed by some other fails.

The isotp.h file is in can-isotp/include/uapi/linux/can/, but when I make can-isotp/linux/can and copy isotp.h into it, it will "make" and it will "make modules_install" but it still won't insmod the can-isotp.ko file, giving the same "Invalid file format" error.

-gran

hartkopp, 2020-01-07:
> So, I went through and retraced my steps, thinking I must have done something incorrectly. When I went for the "make", isotp.c gives me a fatal error: linux/can/isotp.h: No such file or directory
> #include <linux/can/isotp.h> (with pink!)
> compilation terminated.

You need to stay on the top directory! -> see README
Do not go into net/can before starting 'make'.

granmadave, 2020-01-07:
oops- got it.  

The make went fine.  The "make modules_install" gave me errors:
  INSTALL /home/.../can-isotp/net/can/can-isotp.ko
At main.c:160:
-SSL error: 020010002:system library:fopen:No such file or directory: ../crypty/bio/bss_file.c:72
-SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing+key.pem: No such file or directory
  DEPMOD 5.0.0-37-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.

hartkopp, 2020-01-07:
You have a signed kernel in your Ubuntu system and I assume only Ubuntu has the keys to sign modules. Hm - what happens if you try to insmod the module can-isotp.ko directly? Does that work?

granmadave, 2020-01-07:
That is strange.  I didnt think I did a signed kernel.  How would I insmod
directly?

hartkopp, 2020-01-08:
With `insmod can-isotp.ko` :-)
You can try it - but I assume that the kernel would not accept unsigned modules.
What is your Linux hardware? UEFI with secure boot? Or a cloud solution?

pylessard, 2020-01-08:
I am not sure how Ubuntu behave with regards of module signing, but I get the same message under Debian with "make modules_install" I usually insert manually and it works.

When I get "invalid module format" it usually is because I haven't loaded the can module. Doing "sudo modeprobe can" fixes it.

granmadave, 2020-01-08:
I'm doing a vm of xubuntu inside of Windows 10.  I got the xubuntu 32-bit iso directly from xubuntu and didn't specify any strange aspects of it while setting it up.  I've loaded the can module, the vcan module, the slcan module... I keep getting an "invalid module format when trying to insmod the can-isotp.ko module.

hartkopp, 2020-01-09:
@pylessard : I tested your approach with my Debian system too. And I was also able to load can-isotp.ko into the standard (signed) kernel while the kernel was complaining about the unsigned module - but it worked.

@granmadave : Can you please send the output of:
`uname -a` and
`file net/can/can-isotp.ko`
?

JerBcd, 2020-01-15:
> oops- got it.
>
> The make went fine. The "make modules_install" gave me errors:
> INSTALL /home/.../can-isotp/net/can/can-isotp.ko
> At main.c:160:
> -SSL error: 020010002:system library:fopen:No such file [...]

Same SSL error for me, did you find a solution please @granmadave ?

hartkopp, 2020-01-15:
Hi @JerBcd 
the "make modules_install" process fails due to missing crypto keys.
But you should be able to load the module e.g. with
`insmod can-isotp.ko` or `modprobe can-isotp.ko`

granmadave, 2020-02-17:
Sorry about the radio silence; I have been a headless chicken. I have NO idea why it wasn't working. I tried it twice today in different builds and it is working in both. I think maybe it's the "sudo modprobe can" before doing the "insmod ./net/can/can-isotp.ko"?

My only question now is... what's the point of the "-d" flag? Where does it send the data? 

Thank you all! For my initial needs, I consider this item closed.

closed 2020-02-17 by granmadave

hartkopp, 2020-02-17:
-d
refers to which tool?

granmadave, 2020-02-17:
> -d
> refers to which tool?

My apologies. I'm referring here to the isotpsniffer tool, but I'm still confused about its use with isotpdump, too.

hartkopp, 2020-02-17:
All isotp* tools use the same scheme.
You always need TWO CAN-IDs to make a point-to-point ISO-TP connection.
They are given to all of these tools with "-s ID1" and "-d ID2".
But as you **look into the OTHER END** of the connection with **isotprecv** you need to **SWAP** the CAN-IDs only here.
isotpsend -s ID1 -d ID2 can0
isotpdump -s ID1 -d ID2 can0
isotpsniffer -s ID1 -d ID2 can0
isotp**recv** -s **ID2** -d **ID1** can0

granmadave, 2020-02-17:
Interesting. So in the read-out, I shouldn't see anything on ID2, but if I were to do a recv or sniffer with -s ID2, I should see that traffic?

hartkopp, 2020-02-17:
Read my lips:
**"You always need TWO CAN-IDs to make a point-to-point ISO-TP connection."**
When you have a segmented transfer of ISO-TP PDU length of more than 7 bytes you will always see both IDs.
See https://wiki.automotivelinux.org/_media/agl-distro/agl2017-socketcan-print.pdf - slide 29ff

hartkopp, 2020-02-17:
I would strongly recommend to play with the tools to get behind the handling and CAN ID assignments.

granmadave, 2020-02-18:
If I wanted to test what you've got on slide 29 in a VM, would I just use two terminals and in terminal 1 type: ~$echo... | send -s 321 -d 123 vcan0 and in terminal 2 type isotpdump -c -a -s 321 -d 123 vcan0?

...'cause when I do that, I'm only seeing the first byte printed, I'm not getting the response on 123 or any of the other lines on 321. Tx ends, but Rx continues to wait.

hartkopp, 2020-02-20:
Please read https://github.com/hartkopp/can-isotp/issues/20#issuecomment-587132799

hartkopp, 2020-02-20:
Here is the same question answered: https://github.com/hartkopp/can-isotp/issues/15#issuecomment-586693016