hartkopp/can-isotp #21: Backward compatibility for Linux 5.4+ versions in bind()

pylessard, 2020-01-15:
Hi,
I've been flagged an issue with my Python Iso-TP module (https://github.com/pylessard/python-can-isotp/issues/22) regarding a backward compatibility issue. After diving into it, I believe the issue is a conflict between Python and this module.

Apparently, opening an ISOTP socket with Python 3.6.4 and a can-isotp.ko built under Linux 5.4 doesn't work. I believe the problem arise from the addition of J1939 support that modified the size of the ``sockaddr_can`` (change acknowledges by yourself :) ).

Python create the address object for the user and deduce the length of the address object by using ``sizeof(*addr)``. If Python has been compiled with a headers of Linux 5.3, then the this value should be smaller than what one would get with Linux headers of Linux 5.4+.  

This [line here](https://github.com/hartkopp/can-isotp/blob/master/net/can/isotp.c#L1099), checks for conflicts by validating the size of the object. This verification will prevent the creation of ISO-TP socket if an older Python is used with a freshly compiled kernel module.

Is there a way to be backward compatible with this or should one make sure to keep his Python version aligned with his Kernel version?

Thanks!

hartkopp, 2020-01-15:
Fixed in
https://github.com/hartkopp/can-isotp/commit/fbed98a0b5de1bbb29479516cf9b4aeb8b52ec55
and
https://github.com/hartkopp/can-isotp/commit/ced84cab120a1ddde2bd4561a18514707c8a6184
Thanks @pylessard !

closed 2020-01-15 by hartkopp

pylessard, 2020-01-16:
Very interesting, I wasn't aware of the existence of the ``offsetof`` macro! That makes a very nice fix.
Thank you