hartkopp/can-isotp #22: candump shows traffic, isotpdump shows traffic, isotprecv does not receive frame

brandonros, 2020-01-16:
What data can I provide for you?

```
$ candump can0
  can0  7E5   [8]  02 10 03 55 55 55 55 55
  can0  7E5   [8]  02 10 03 55 55 55 55 55
  can0  7E5   [8]  02 10 03 55 55 55 55 55
  can0  7E5   [8]  02 10 03 55 55 55 55 55
```

```
$ isotpdump -s 7e5 -d 7ed can0
 can0  7E5  [8]  [SF] ln: 2    data: 10 03 55 55 55 55 55 
```

```
$ isotprecv -s 7e5 -d 7ed can0

```

never returns a frame, no matter how many times I resend it (yet it still shows up in isotpdump + candump while isotprecv is running + waiting)

hartkopp, 2020-01-16:
This is weird. In fact I started isotpsniffer in parallel which also uses the can-isotp module and it worked!

`$ isotpsniffer -s 7e5 -d 7ed vcan0`
`vcan0  7E5  [2]  10 03  - '..'`
`vcan0  7E5  [2]  10 03  - '..'`

I'll take a look at it todays afternoon ...

hartkopp, 2020-01-16:
Ah found it!
You have to swap (!!) source and destination CAN IDs on isotprecv - as you need to look from the other side of the "virtual point-to-point' channel.
So 
`isotprecv -d 7e5 -s 7ed vcan0`
makes it :-)

closed 2020-01-16 by hartkopp