hartkopp/can-isotp #24: insmod: ERROR: could not insert module ./net/can/can-isotp.ko: Unknown symbol in module

litaoxyz, 2020-02-14:
When I sudo insmod ./net/can/can-isotp.ko ,terminal print “insmod: ERROR: could not insert module ./net/can/can-isotp.ko: Unknown symbol in module”.
The results of "dmesg | tail -n 1" is "can_isotp: Unknown symbol can_rx_register (err -2)".
I want to load this on raspiberry pi 3b+.

The result of "uname -a" is "Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux".
The result of "file net/can/can-isotp.ko" is "net/can/can-isotp.ko: ELF 32-bit LSB relocatable, ARM, EABI5 version 1 (SYSV), BuildID[sha1]=f52ef777e54005ea9204e616f0fb5a7335f543a8, not stripped".
What should I do to solve this problem ? thanks.

hartkopp, 2020-02-14:
When using insmod the kernel loads the can-isotp module without taking care about dependencies.
Try 'modprobe' instead to make sure the can.ko module is automatically loaded before.
Or run 'modprobe can-raw' which would load can.ko and can-raw.ko in advance.
can.ko contains the can_rx_register symbol.

litaoxyz, 2020-02-14:
> When using insmod the kernel loads the can-isotp module without taking care about dependencies.
> Try 'modprobe' instead to make sure the can.ko module is automatically loaded before.
> Or run 'modprobe [...]

Thanks a lot. After I loaded can.ko by using 'modprobe can-raw',problem solved.

closed 2020-02-14 by litaoxyz