hartkopp/can-isotp #31: How to fix OSError: [Errno 97] on Jetson TX2?

kilianbr, 2020-09-01:
Hi everyone, 
I'm trying to use this library on a Jetson TX2 Board (arm architecture if that matters) inside of an docker container.

If I test the following code from the docs: 

```
import isotp

s = isotp.socket()
s2 = isotp.socket()
# Configuring the sockets.
s.set_fc_opts(stmin=5, bs=10)

s.bind("vcan0", isotp.Address(rxid=0x123 txid=0x456))
s2.bind("vcan0", isotp.Address(rxid=0x456, txid=0x123))
s2.send(b"Hello, this is a long payload sent in small chunks of 8 bytes.")
print(s.recv())
```

I get the following error message: 

> root@d1a297bae985:/# python3.7 test.py
> Traceback (most recent call last):
>   File "test.py", line 3, in <module>
>     s = isotp.socket()
>   File "/usr/local/lib/python3.7/site-packages/isotp/tpsock/__init__.py", line 74, in __init__
>     self._socket = socket_module.socket(socket_module.AF_CAN, socket_module.SOCK_DGRAM, socket_module.CAN_ISOTP)
>   File "/usr/local/lib/python3.7/socket.py", line 151, in __init__
>     _socket.socket.__init__(self, family, type, proto, fileno)
> OSError: [Errno 97] Address family not supported by protocol

I already searched the internet but didn't find anything helping me in this context. I am not sure why it isn't working (it worked before on a RPI3). Ifconfig shows both interfaces, can0 and can1. 

I would really appreciate any help!

hartkopp, 2020-09-01:
Hi,
when you are using the Python ISO-TP package from https://github.com/pylessard/python-can-isotp you can either use the user space based application or the kernel based implementation with the ISO-TP sockets.
Your problem description shows that you try to access the kernel based implementation which requires a kernel module built from the code in this repository -> see https://github.com/hartkopp/can-isotp/blob/master/README.isotp

pylessard, 2020-09-01:
In his previous issue, @kilianbr mentioned that he did built the kernel module. Also, the error he would get if no kernel module would have been loaded would be "Protocol not supported".  This error is something else. It almost looks like python doesn't support the protocol, but it should as it seems to be  Python 3.7.

@kilianbr  : Can you try to send data with isotpsend and ``isotprecv`` to ``discriminate`` Python?

kilianbr, 2020-09-02:
Hi, first of all thanks for your replies. I think it has to do something with my kernel module. I have to use a yocto dunfell OS and to be honest, know only very little about that. Unfortunately I don't have access onto my device until next week and therefore can't try the suggested commands. I will try it and reply as soon as possible next week! Best regards

hartkopp, 2020-10-19:
Hi @kilianbr,
any update on this issue?

kilianbr, 2020-10-19:
Hi @hartkopp, sorry that I didn't answer. It turned out that my organization didn't implement the CAN on the yocto image and therefore I wasn't able to open a socket. 
So it wasn't any error with the library but with my hardware. 
Thanks again for the replies and all the best!

closed 2020-10-19 by kilianbr