hartkopp/can-isotp #33: #include "compat.h" obsolete?

Thomseeen, 2020-09-30:
Searching for compatibility issues I have (probably on my side may come back later if I can't resolve them :D), I found this `#include "compat.h"` in `isotp.c` which seems to be something from the old repository but isn't needed anymore?!

hartkopp, 2020-09-30:
This compat.h is only pulled on Linux Kernels < 2.6.25
https://github.com/hartkopp/can-isotp/blob/master/net/can/isotp.c#L80
So it shouldn't affect your build process - or for which kernel version do you plan to build the code??

Thomseeen, 2020-09-30:
I am trying to include the module in a buildroot environment with Linux 2.6.31 which ends up failing because of a few missing defines in supposedly `linux/can.h`. This is quite the sudden deep dive into linux-stuff, so I was just looking around noticing that include without finding the actual file for it in this repo.

hartkopp, 2020-09-30:
Yes. The latest version of ISO-TP supports also CAN FD which was introduced in Linux 3.6. So the implementation requires these definition from can.h even if you can't use CAN FD with your 2.6.31 kernel.
I backported the entire CAN stuff for a 2.6.13 kernel a *long* time ago :-D
It should work when you update some of theses include bits like can.h.

The CAN FD support has been added in the (really obsolete) can-isotp-modules repository here:
https://github.com/hartkopp/can-isotp-modules/commit/b990ca64178bd4ead1e14879b35b29fb50066a33

So if you can't get it to work, you might also take a look at that ancient code (but not really recommended).

Thomseeen, 2020-10-01:
So... I managed to include a patched can.h-header file in my buildroot build, but I still get an error about a `can_ioctl` function. Any hints on that? :)

hartkopp, 2020-10-01:
Can you check the issue with this updated check for your kernel version?

Thomseeen, 2020-10-01:
Yes, seems like it wasn't even needed... works like a charm now. Thanks a lot!

hartkopp, 2020-10-01:
Have fun!

closed 2020-10-01 by hartkopp