hartkopp/can-isotp #38: ISSUE: ISO-TP Kernel build on Raspberry-Pi 4B

Kallappambk, 2020-12-11:
Hi, 

I am trying to build CAN-ISOTP module by refering the link https://github.com/hartkopp/can-isotp and facing problem with installing Kernel headers 
Error log for command : **sudo apt-get install build-essential linux-headers-$(uname -r)**

pi@raspberrypi:~/ISOTP/can-isotp $ sudo apt-get install build-essential linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-5.4.44-v7l
E: Couldn't find any package by glob 'linux-headers-5.4.44-v7l'
E: Couldn't find any package by regex 'linux-headers-5.4.44-v7l'

kindly suggest how to build for RPI because i am new  to these things

Thanks in advance

hartkopp, 2020-12-11:
Hi,

this is not a can-isotp issue.
Please take a look at this tread: https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4#issuecomment-641287505

closed 2020-12-11 by hartkopp

Kallappambk, 2020-12-14:
thanks for the responce.
able to fix this issue.