hartkopp/can-isotp #40: Can't insert kernel module because of version issue

kilianbr, 2021-01-05:
Hi, 
I am using your library for quite a time and it was working fine until yesterday. I already opened an issue here: https://github.com/pylessard/python-can-isotp/issues/37

I thought that I inserted the kernel module as described in the description the correct way (and I did since it was working fine). Now I faced the mentioned issue and discovered that I get an error inserting the module:

[image]

After using 

`dmesg`

I can see that there is a problem with my kernel version:

[image]

I already did the standard procedure

`sudo apt-get update & 
 sudo apt-get upgrade`

And tried to manually get the correct version from https://packages.ubuntu.com/bionic/linux-image-generic and updated my  ubuntu version from 18.04 to 20 but its still not working.

Do you have any suggestions how to solve this?

Best regards,

hartkopp, 2021-01-05:
You need to compile the can-isotp module again after changing the Linux kernel it should run in.
This is not a library but a kernel modules which relies on the environment of that specific kernel version it was compiled for.

kilianbr, 2021-01-05:
Thanks for the answer. I tried to compile the module again but it results in an ssl error. I tried to look it up but did not find any 
useful information about it.
[image]

If I try to ignore it like described here: https://debianforum.de/forum/viewtopic.php?t=178945, inserting the module will fail with the following error logs:

[image]

hartkopp, 2021-01-05:
Oh, you are almost done!!

As auto loading does not work for out-of-tree modules, you need to run
`modprobe can`
before.
This provides the required symbols.

closed 2021-01-06 by kilianbr

hartkopp, 2021-01-06:
`can.ko` is no CAN driver. It provides the network layer infrastructure for CAN protocol implementations like CAN_RAW, CAN_BCM, CAN_J1939 and CAN_ISOTP.
OTOH `can-dev.ko` provides the CAN driver layer infrastructure.