hartkopp/can-isotp #50: how to

jinbaotang, 2021-12-20:


jinbaotang, 2021-12-20:
[image]
I want to use cantp data length > 4095, but I read the isotp.h, cannot find the optional to set?is isotp support it?

hartkopp, 2021-12-20:
Hi,
to be able to test this feature I set the max. PDU length to 8200, see here:
https://github.com/hartkopp/can-isotp/blob/master/net/can/isotp.c#L110
https://github.com/torvalds/linux/blob/master/net/can/isotp.c#L86

I have no final idea what needs to be done, when users *really* want to transfer 4Gbyte PDUs - but even with the current implementation you can transfer e.g. 8000 byte PDUs which should automatically switch to the FF_DL representation depicted in your request.

Please let me know when you run into problems with it. Thanks!

jinbaotang, 2021-12-22:
> Hi, to be able to test this feature I set the max. PDU length to 8200, see here: https://github.com/hartkopp/can-isotp/blob/master/net/can/isotp.c#L110 [...]

thank harkopp, I got it

closed 2021-12-22 by jinbaotang