hartkopp/can-isotp #59: MSG_CMSG_COMPAT set by the kernel for `recvmsg`

dissecto, 2023-03-30:
Hi, not sure if this is the right place for this issue, but I've encountered a problem with more recent kernels > 5.10 on ARM64 machines. 

On any recvmsg call from the kernel, the flag MSG_CMSG_COMPAT is set on my machine, which results in EINVAL on this line: 
https://github.com/hartkopp/can-isotp/blob/7626d0a0707391970080d493ce69638719938da7/net/can/isotp.c#L1057

The issue can be reproduced on any Raspberry PI 4 with the latest kernel. 

One remark, `isotprecv` is not affected by this issue, since it uses `read` and not `recvmsg`. 

I propose to add this flag to this line: https://github.com/hartkopp/can-isotp/blob/7626d0a0707391970080d493ce69638719938da7/net/can/isotp.c#L1057

marckleinebudde, 2023-04-05:
> Hi, not sure if this is the right place for this issue, but I've encountered a problem with more recent kernels > 5.10 on ARM64 machines.
> 
> On any recvmsg call from the kernel, the flag MSG_CMSG_COMPAT is set on my machine, which results in EINVAL on this line:

Are you using a 32 bit userspace?

dissecto, 2023-04-05:
Yes, I‘m using a 32bit userspace.

marckleinebudde, 2023-04-05:
> Yes, I‘m using a 32bit userspace.

This is where the `MSG_CMSG_COMPAT` comes from.

marckleinebudde, 2023-04-05:
> @marckleinebudde @olerem : Using `MSG_CMSG_COMPAT` in recvmsg flags seems to be a common pattern, e.g. in af_packet.c or bluetooth/hci_sock.c .
> 
> I don't see any 'special' handlings in cmsg's in isotp.c but some special created cmsg's in j1939/socket.c (search for `put_cmsg()`).

Looks good send a patch.