hartkopp/can-isotp #60: Compiling of branch mainline-5.4+

pompushko, 2023-11-01:
Hello

I tried to test new branch on Ubuntu 23.10 with 6.5.0-10-generic x86_64 kernel

```
user@canbus-dev:~/can-isotp$ make
make -C /lib/modules/6.5.0-10-generic/build M=/home/user/can-isotp modules
make[1]: Entering directory '/usr/src/linux-headers-6.5.0-10-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  CC [M]  /home/user/can-isotp/net/can/isotp.o
  LD [M]  /home/user/can-isotp/net/can/can-isotp.o
  MODPOST /home/user/can-isotp/Module.symvers
  CC [M]  /home/user/can-isotp/net/can/can-isotp.mod.o
  LD [M]  /home/user/can-isotp/net/can/can-isotp.ko
  BTF [M] /home/user/can-isotp/net/can/can-isotp.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-10-generic'
```

```
user@canbus-dev:~/can-isotp$ sudo make modules_install
make -C /lib/modules/6.5.0-10-generic/build M=/home/user/can-isotp modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.5.0-10-generic'
  INSTALL /lib/modules/6.5.0-10-generic/updates/net/can/can-isotp.ko
  SIGN    /lib/modules/6.5.0-10-generic/updates/net/can/can-isotp.ko
  DEPMOD  /lib/modules/6.5.0-10-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-10-generic'
```

And when I trying to insert new module:

```
user@canbus-dev:~/can-isotp$ sudo insmod ./net/can/can-isotp.ko
insmod: ERROR: could not insert module ./net/can/can-isotp.ko: Unknown symbol in module
```

Thank you

hartkopp, 2023-11-01:
`insmod` does not necessarily load depending modules like `can.ko`

Please run `modprobe can` before the `insmod`

pompushko, 2023-11-01:
Yeah. Indeed. That helped me.. Thank you!

Also, how can I load that module during system startup?

closed 2023-11-03 by hartkopp