hartkopp/can-isotp #62: Address extension in one direction but not the other

pylessard, 2024-01-03:
I was implementing a feature requested by a user on the python-can-isotp module where one can send using a 11 bits ID and receive with a 29 bits ID. I generalized the idea and let the user sends with any kind of ISOTP address and receive with any kind of ISOTP address.

By doing so, I noticed that the isotp socket cannot use an address extension in a direction without using one in the other direction.  I do not have any use case for that scenario, but I wonder if it should be allowed since it would be the most flexible solution.

Have a look at [what I was hoping to achieve](https://github.com/pylessard/python-can-isotp/pull/107/files#diff-f548dd1b8e86f179d210f52d4549876cea586e0247ee5671bb9f3b81386ee8c0R264-R267)

But I had to[ add this check ](https://github.com/pylessard/python-can-isotp/pull/107/files#diff-f548dd1b8e86f179d210f52d4549876cea586e0247ee5671bb9f3b81386ee8c0R239-R241)

Any opinion on this?

hartkopp, 2024-01-03:
> I was implementing a feature requested by a user on the python-can-isotp module where one can send using a 11 bits ID and receive with a 29 bits ID.

I had to double check this myself - but the kernel implementation already supports this 11/29 bit mix.

Interesting that there are really people working with such weird setups.

> I generalized the idea and let the user sends with any kind of ISOTP address and receive with any kind of ISOTP address.
>
> By doing so, I noticed that the isotp socket cannot use an address extension [...]

So far I've seen only two use cases that use the address extension. And in these use cases it is just clear that AE is enabled or not (on the connection level).

> Have a look at [what I was hoping to achieve](https://github.com/pylessard/python-can-isotp/pull/107/files#diff-f548dd1b8e86f179d210f52d4549876cea586e0247ee5671bb9f3b81386ee8c0R264-R267)
>
> But I had [...]

Of course it would be no big deal to adapt the kernel code in that way - but I checked the options and have the feeling that the sockopt API with additional flags would become even more confusing for the user then.

As I also do not see a real use case for this feature right now I would tend to raise the discussion again, when someone really shows up with this error message in his hands:
`"The IsoTP socket module does not support asymmetric addresses with inconsistent address_extension byte"` 
;-)

My assumption is that no one will ever show up with it  `¯\_(ツ)_/¯`

pylessard, 2024-01-03:
that's what I believe too. I guess we should close this issue. I will add a comment that points to it in my code base.

Thanks for your input

closed 2024-01-03 by pylessard