hartkopp/can-isotp #64: read: Network is down

pompushko, 2024-04-11:
Hello

I'm using @pylessard **udsoncan** python library and have some strange workaround after read_data_by_identifier and not big answer.

Here is part of code, after which I got that error during candump:
```
response_F190 = client.read_data_by_identifier(0xF190)
response_F1A0 = client.read_data_by_identifier(0xF1A0)
```

Here is candump output during executing this request
```
  can0  18DA20F1   [3]  02 10 03
  can0  18DAF120   [7]  06 50 03 00 32 01 F4
  can0  18DA20F1   [4]  03 22 F1 90
  can0  18DAF120   [8]  10 14 62 F1 90 5A 41 52
  can0  18DA20F1   [3]  30 00 00
  can0  18DAF120   [8]  21 46 41 45 41 00 00 00
  can0  18DAF120   [8]  22 37 35 36 37 00 00 00
  can0  18DA20F1   [4]  03 22 F1 A0
  can0  18DAF120   [8]  10 43 62 F1 A0 35 30 35
  can0  18DA20F1   [3]  30 00 00
  can0  18DAF120   [8]  21 33 34 30 35 34 20 20
  can0  18DAF120   [8]  22 20 5A 41 52 46 41 45
  can0  18DAF120   [8]  23 41 56 36 4A 00 35 36
  can0  18DAF120   [8]  24 37 00 00 00 00 00 00
  can0  18DAF120   [8]  25 4D 00 30 00 20 20 20
  can0  18DAF120   [8]  26 20 01 30 30 5F 30 00
  can0  18DAF120   [8]  27 5F 30 35 5F 30 00 00
  can0  18DAF120   [8]  28 01 20 20 20 20 20 20
  can0  18DAF120   [6]  29 00 52 50 00 00
read: Network is down
```

Some time ago it works good. But now it start again 🗡️ 

```
can@canbus:~ $ uname -a
Linux canbus6.1.0-rpi8-rpi-v7 #1 SMP Raspbian 1:6.1.73-1+rpt1 (2024-01-25) armv7l GNU/Linux
```

```
can@canbus:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
```

Thank you!

hartkopp, 2024-04-12:
Hi @pompushko 
the isotp implementation in the Linux kernel is not able to shut down CAN interfaces.
Is it possible that your application or the script starting your application sets the CAN interface parameters - and later shuts down the interface?

Can you post the output of `ip -det link show can0` before and after the described `read: Network is down` issue?

pompushko, 2024-04-12:
Seems like something wrong with my CAN transceiver....

closed 2024-04-12 by pompushko