hartkopp/can-isotp #67: make fails and I can't figure out why. Hoping someone can help out

id74em8, 2024-07-19:
on running make, this is the output

make -C /lib/modules/6.8.0-36-generic/build M=/home/pentester/Downloads/can-isotp modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-36-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/pentester/Downloads/can-isotp/net/can/isotp.o
/home/pentester/Downloads/can-isotp/net/can/isotp.c:93:2: error: #error No need to compile this out-of-tree driver! ISO-TP is part of Linux Mainline kernel since Linux 5.10.
   93 | #error No need to compile this out-of-tree driver! ISO-TP is part of Linux Mainline kernel since Linux 5.10.
      |  ^~~~~
/home/pentester/Downloads/can-isotp/net/can/isotp.c: In function ‘isotp_recvmsg’:
/home/pentester/Downloads/can-isotp/net/can/isotp.c:1064:44: warning: passing argument 3 of ‘skb_recv_datagram’ makes pointer from integer without a cast [-Wint-conversion]
 1064 |         skb = skb_recv_datagram(sk, flags, noblock, &ret);
      |                                            ^~~~~~~
      |                                            |
      |                                            int
In file included from ./include/net/net_namespace.h:43,
                 from ./include/linux/netdevice.h:38,
                 from /home/pentester/Downloads/can-isotp/net/can/isotp.c:60:
./include/linux/skbuff.h:3972:77: note: expected ‘int *’ but argument is of type ‘int’
 3972 | struct sk_buff *skb_recv_datagram(struct sock *sk, unsigned int flags, int *err);
      |                                                                        ~~~~~^~~
/home/pentester/Downloads/can-isotp/net/can/isotp.c:1064:15: error: too many arguments to function ‘skb_recv_datagram’
 1064 |         skb = skb_recv_datagram(sk, flags, noblock, &ret);
      |               ^~~~~~~~~~~~~~~~~
./include/linux/skbuff.h:3972:17: note: declared here
 3972 | struct sk_buff *skb_recv_datagram(struct sock *sk, unsigned int flags, int *err);
      |                 ^~~~~~~~~~~~~~~~~
/home/pentester/Downloads/can-isotp/net/can/isotp.c: At top level:
/home/pentester/Downloads/can-isotp/net/can/isotp.c:1630:10: error: ‘const struct proto_ops’ has no member named ‘sendpage’
 1630 |         .sendpage = sock_no_sendpage,
      |          ^~~~~~~~
/home/pentester/Downloads/can-isotp/net/can/isotp.c:1630:21: error: ‘sock_no_sendpage’ undeclared here (not in a function); did you mean ‘sock_no_sendmsg’?
 1630 |         .sendpage = sock_no_sendpage,
      |                     ^~~~~~~~~~~~~~~~
      |                     sock_no_sendmsg
make[4]: *** [scripts/Makefile.build:243: /home/pentester/Downloads/can-isotp/net/can/isotp.o] Error 1
make[3]: *** [scripts/Makefile.build:481: /home/pentester/Downloads/can-isotp/net/can] Error 2
make[2]: *** [/usr/src/linux-headers-6.8.0-36-generic/Makefile:1926: /home/pentester/Downloads/can-isotp] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-36-generic'
make: *** [Makefile:21: modules] Error 2

HenrikWestphal, 2024-07-24:
try the mainline-5.4+ Branch 
git clone -b mainline-5.4+ --single-branch https://github.com/hartkopp/can-isotp.git

hartkopp, 2024-07-25:
PLEASE read the  README!!

"Linux Kernel Module for ISO 15765-2:2016 CAN transport protocol PLEASE NOTE: This module is part of the mainline Linux kernel since version 5.10 "

You don't need to compile this module for your 6.8 kernel. It is already there and working.

hartkopp, 2024-07-28:
Ok, as the compiler output already solved this issue it can be closed.

```
make -C /lib/modules/6.8.0-36-generic/build M=/home/pentester/Downloads/can-isotp modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-36-generic'
warning: the compiler differs from the one used to build the kernel
The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
You are using: gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
CC [M] /home/pentester/Downloads/can-isotp/net/can/isotp.o
/home/pentester/Downloads/can-isotp/net/can/isotp.c:93:2: error: #error No need to compile this out-of-tree driver! ISO-TP is part of Linux Mainline kernel since Linux 5.10.
93 | #error No need to compile this out-of-tree driver! ISO-TP is part of Linux Mainline kernel since Linux 5.10.
| ^~~~~
```

closed 2024-07-28 by hartkopp