hartkopp/can-isotp #68: CAN you add CANFD support in isotp-c, because the can is not only support can2.0b 8 bytes, it also need CANFD 64 bytes support in the pure electronic vehicle, can you add CANFD support?

zhangjingliu, 2025-01-13:
CAN you add CANFD support in isotp-c, because the can is not only support can2.0b 8 bytes, it also need CANFD 64 bytes support in the pure electronic vehicle, can you add CANFD support?

hartkopp, 2025-01-14:
This is already implemented.

You need to set the can_isotp_ll_options accordingly to enable CAN FD.

closed 2025-01-14 by hartkopp

zhangjingliu, 2025-01-14:
> This is already implemented.
> 
> You need to set the can_isotp_ll_options accordingly to enable CAN FD.
> 
> Edit: Please take a look into isotpsend.c and isotprecv.c to see how it works.

thank you, brother, have a good day

pylessard, 2025-01-14:
This module is a Linux driver. It's not supported on Windows. There are other projects you can use. 
If you can use Python, I have a python implementation that can work on top of python-can, that works on Windows and support lots of hardware

zhangjingliu, 2025-01-14:
> This module is a Linux driver. It's not supported on Windows. There are other projects you can use. If you can use Python, I have a python implementation that can work on top of python-can, that works on Windows and support lots of hardware

but I just need MCU embedded isotp module, and I have see you python-can , it is a PC software.
I mean I'm use a STM32 or ESP32 or NXP MCU, I need to port hartkopp's can-isotp module.
I need to use STM32 timer and can recieve IRQ interrupt to recieve can frame message and send. not need the PC upper python software

pylessard, 2025-01-14:
My opinion is that you should find another project to port to your mcu. Kernel driver and baremetal code are very different. I eve  think you would have less trouble to convert my python code to C/C++

zhangjingliu, 2025-01-14:
> My opinion is that you should find another project to port to your mcu. Kernel driver and baremetal code are very different. I eve think you would have less trouble to convert my python code to C/C++

it that easy work, thank you very much！I need to read your python code

hartkopp, 2025-01-15:
> My opinion is that you should find another project to port to your mcu. Kernel driver and baremetal code are very different. I eve think you would have less trouble to convert my python code to C/C++

The entry points from the kernel API to the handling of CAN frames for ISO-TP are pretty easy to understand IMO.
If you read and understand the code you can easily pick some code or concepts.

In fact the kernel code provides many additional features also for testing or for use-cases that are not even covered by the ISO15765-2 standard. So understanding the code and picking the really relevant features for the embedded implementation would be the main task.

Maybe there are other ISOTP implementations around that are more easily to understand.

pylessard, 2025-01-15:
https://github.com/lishen2/isotp-c/

There's this. I don't see canfd support though

zhangjingliu, 2025-01-15:
> https://github.com/lishen2/isotp-c/
> 
> There's this. I don't see canfd support though

I have see this lishen2's code, but he don't know CANFD, he just know can 8 byte, so I give up to use her module

zhangjingliu, 2025-01-15:
Actually, CANFD is different from CAN2.0B , except the SF,FF, the other CF and FC is the same

zhangjingliu, 2025-01-15:
> > https://github.com/lishen2/isotp-c/
> > There's this. I don't see canfd support though
> 
> I have see this lishen2's code, but he don't know CANFD, he just know can 8 byte, so I give up to use her module

and I see the lishen2's code layer, and find that his isotp use the union and struct to decode the can SF,FF,FC,CF, so if I add CANFD, lishen2's code need much more change the union can frame struct

zhangjingliu, 2025-01-15:
if I change lishen2's can to CANFD, it will be like this 
[image]

zhangjingliu, 2025-01-15:
also need to define CANFD decode in the SF and FF recieve and send,  the FC and CF remain

zhangjingliu, 2025-01-15:
> > https://github.com/lishen2/isotp-c/
> > 就是这样。不过我没有看到 canfd 支持
> 
> 我看到了这个 lishen2 的代码，但是他不懂 CANFD，他只知道 can 8 byte，所以我放弃了使用她的模块
the CAN2.0B
[image]
the CANFD as follows
[image]

zhangjingliu, 2025-01-15:
[image]

zhangjingliu, 2025-01-15:
> The kernel implementation has been created during the ISO15765-2 standardization process.
> 
> You can check the changes that have been made for CAN FD in this outdated repository: https://github.com/hartkopp/can-isotp-modules/commits/master/?after=314ea73e97934a1f0f0d13404e9ec4f979fb1b35+9

Yes, I see ,your project is support CAN/CANFD both , it is an perfect CAN ISOTP module, and could you can port this module to a STM32 or ESP32 MCU platform , since you have it in linux driver

zhangjingliu, 2025-01-15:
and my platform is MCU chip, like Infineon TC275/ TLE9855, STM32, NXP S32K144. I'm not use linux usually and not family with linux. but I have see your ISOTP module , it is very good.

hartkopp, 2025-01-15:
> Yes, I see ,your project is support CAN/CANFD both , it is an perfect CAN ISOTP module, and could you can port this module to a STM32 or ESP32 MCU platform , since you have it in linux driver

I think the ISO15765-2 standard is pretty good to understand. And additionally you have my implementation that also uses a similar naming. I don't have such hardware here and also no use-case for such an implementation.

You can do that for sure.

zhangjingliu, 2025-01-15:
ok, it will be a little difficult for me to port this to STM32 , mainly is the timer handle ,is different from the linux timer